esp-idf-lib/core #29: Porting VL53L1X driver

trombik, 2025-08-01:
### Device name

VL53L1X

### Device description

VL53L1X are Time-of-Flight (ToF) and  laser-ranging sensor from ST Electronics. 

### Device product page URL

https://www.st.com/en/imaging-and-photonics-solutions/vl53l1x.html

### Datasheet URL

https://www.st.com/resource/en/datasheet/vl53l1x.pdf

### Can the device be purchased at any of the followings?

- [X] AliExpress
- [X] Amazon
- [X] ebay
- [X] Digi-Key
- [X] Mouser

### URLs to purchase the device

https://www.amazon.com/CQRobot-Ocean-Time-Flight-Compatible/dp/B07F3TV3G4/ref=sr_1_2?crid=31KO9NORW6DK3&keywords=vl53l1x&qid=1697483150&sprefix=vl53l%2Caps%2C492&sr=8-2

### Other implementations

https://github.com/leeebo/esp32-vl53l1x-test/tree/master

trombik, 2025-08-01:
the original issue: https://github.com/UncleRus/esp-idf-lib/issues/581

UncleRus, 2026-11-14:
Duplicate of #22

closed 2026-11-14 by UncleRus